duinoapp/duinoapp-client #7: Esp32 support?

jsyzthz, 2021-01-18:
auinoapp is a good job. it can online compile and upload without third tools.  
i saw esp8266. support esp32？

mrfrase3, 2021-01-20:
not too sure about the esp32, I haven't got one to experiment with. I will get my hands on one and look further into this.

madshobye, 2021-02-24:
When compiling to esp32 I get the following error: no module named serial.
It seems to be a problem that the compile servers are missing a version of python:
https://askubuntu.com/questions/320996/how-to-make-python-program-command-execute-python-3/1031733#1031733
https://github.com/espressif/arduino-esp32/issues/13
(i cannot copy paste from the prompt - for some strange reason)

mrfrase3, 2021-08-14:
Howdy,

Long time, no update. How are you?

Anyway, after a fair bit of effort, I've implemented a propper solution to ESP programming, which is now available in the 3.3.1 beta version of the app, beta.duino.app 

Feel free to give it a go, let me know if there are any issues.