duinoapp/duinoapp-client #16: Serial input issue

Hesstre, 2021-02-18:
Hey! Im trying to use chromeduino 3 with my  arduino nano, and im getting some issues with the serial input. My arduino/USB is getting dected and I can click connect and pair it but it  doesent seem to affect anything. I checked on my pc and all the settings in chromduino are the same as the Arduino IDE ones:( arduino nano and ATmega 328p (old bootloader)) so im not sure whats the problem. If anyone could help that would be great

mrfrase3, 2021-03-02:
Have you previously been successful at programming that Arduino with the Arduino IDE?

mrfrase3, 2021-05-04:
Hey, just pinging you on this, otherwise, I'll close this ticket. (maybe try not the old bootloader)

closed 2021-05-08 by mrfrase3