duinoapp/duinoapp-client #22: Uploading code stuck at 50%

shankinsecs, 2021-04-21:
We currently use Chromeduino at our school and would like to transition to the Duino App. Our students and facutly utilize chromebooks. We have attempted to upload code to an Arduino Uno from several different devices and continually the upload gets stuck at 50%. Here is the output we receive in the serial box. 

Using board 'uno' from platform in folder: /mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.3
Using core 'arduino' from platform in folder: /mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.3
Detecting libraries used...
/mnt/duino-data/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/standard /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/sketch/test_2.ino.cpp -o /dev/null
Generating function prototypes...
/mnt/duino-data/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/standard /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/sketch/test_2.ino.cpp -o /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/preproc/ctags_target_for_gcc_minus_e.cpp
/mnt/duino-data/.arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/mnt/duino-data/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/standard /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/sketch/test_2.ino.cpp -o /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/sketch/test_2.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /tmp/arduino-core-cache/core_arduino_avr_uno_ede048c5b12cee71cbf4bcf1685205d6.a
Linking everything together...
/mnt/duino-data/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/test_2.ino.elf /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/sketch/test_2.ino.cpp.o /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/../arduino-core-cache/core_arduino_avr_uno_ede048c5b12cee71cbf4bcf1685205d6.a -L/tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A -lm
/mnt/duino-data/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/test_2.ino.elf /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/test_2.ino.eep
/mnt/duino-data/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/test_2.ino.elf /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/test_2.ino.hex
/mnt/duino-data/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size -A /tmp/arduino-sketch-000B1D9FAF37731719533F163A25686A/test_2.ino.elf
Sketch uses 1986 bytes (6%) of program storage space. Maximum is 32256 bytes.
Global variables use 196 bytes (9%) of dynamic memory, leaving 1852 bytes for local variables. Maximum is 2048 bytes.

mrfrase3, 2021-04-21:
Hey there,

All that log is showing is that it compiled fine on the server. If you right-click on the application, select "inspect" a debug menu should open, navigate to the "console" tab.

Try to upload your code see if any errors appear in red. You can click on the errors to expand them out. If there are any, then could you upload a screenshot?

mrfrase3, 2021-04-21:
Also, just a sanity check, have you tried disconnecting and reconnecting the device, and if that didn't work, rebooting the computer?

shankinsecs, 2021-04-22:
[image]
We definitely have. We have also tried different boards and different devices. I do not see any errors in the console tab of the developer tools. Here is a screenshot.

MrWollenburg, 2021-04-23:
Our school is having the same issue. 50% and it will not load beyond. We use Lenovo 500e Chromebooks. I have access to a windows machine with Arduino software on it and the code loads fine on it. Duino 2 always seemed to work, hoping this can be fixed.   :)

shankinsecs, 2021-04-23:
We have tried on HP G5 G6 and G7 chromebooks, as well as a pixelbook and the Dell 3100 EE chromebooks. Chromeduino 2 was working out for us previously as well but it no longer loads. Just a white screen.

MrWollenburg, 2021-04-23:
Shankinsecs, a solution I'm using for my students right now is the following:

1.Navigate to https://edu.codebender.cc/class_demo
2. Copy student code.
3. Paste student code into URL bar
4. Copy code from URL (this puts it all into a single line of code, rather than multiple lines)
5. Delete code from codebender
5. Paste code into codebender
6. Upload code

[image]

Hopefully this helps your school as a temp fix!

mrfrase3, 2021-04-23:
Hmmm, I think this might be the timing issue. We've just gone into a Covid lockdown for the weekend where I live so I now have plenty of time to look into this.

Side note, I'm not a huge fan of code bender after they told me to fix their code for free which they charge people for. You should still be able to use chromeduino 2 until June or something, hard to get an exact date...

MrWollenburg, 2021-04-23:
I'd love to use Duino 2, but unfortunately, we can't get passed the server list screen any more.

shankinsecs, 2021-04-23:
> I'd love to use Duino 2, but unfortunately, we can't get passed the server list screen any more.

Same issue for us. Students who had previously used it are able to. However when I reset their settings and clear the cache it no longer works. We also aren't able to get it to work for new students who hadn't previously used it. Hopefully this information is helpful. We appreciate your time so much mrfrase3!

mrfrase3, 2021-04-23:
I found some of the older servers were timing out which paused the loading of the servers until they were all loaded/timeout. try just waiting on the empty servers list for a few minutes.

shankinsecs, 2021-04-23:
Unfortunately no matter how long I leave it sitting on any of our devices I am only getting a blank screen. [image]

mrfrase3, 2021-04-24:
hm, I'd like to say that surprises me, but chromduino 2 has been so flaky that I'm not...

Anyway, I think I've fixed this, and I've staged a 3.2.1 update with it in [beta.duino.app](https://beta.duino.app). If someone could test and confirm that its fixed in beta, I'll deploy it into production.

shankinsecs, 2021-04-24:
Still getting hung at 50% for me. Not getting any errors in the dev console but I´m going to keep trying and I´ll let you know.

mrfrase3, 2021-04-25:
looks like the fix wasn't actually applied, I'm deploying version 3.2.2 now to beta. Sorry about that

shankinsecs, 2021-04-25:
It appears to be working fine for us now. Thank you so much for your time. Our teachers will be thrilled.

closed 2021-04-25 by shankinsecs

mrfrase3, 2021-04-25:
Cool, I've merged beta into production, you should now be able to use version 3.2.2 on https://duino.app/

Enjoy!

MrWollenburg, 2021-04-26:
> Cool, I've merged beta into production, you should now be able to use version 3.2.2 on https://duino.app/
> 
> Enjoy!

The app works great again, thank you for your hard work! Hope the temp lockdown doesn't cause too much fuss with you guys.