duinoapp/duinoapp-client #23: Cannot add servers

MrWollenburg, 2021-04-23:
I tried added a few of the old Duino 2 servers and it appears none of them can connect, even though their URLs are active.

[image]

mrfrase3, 2021-04-23:
Hey,

With a change in protocol, the old servers from chromeduino 2 are not compatible with version 3 (new v3 servers should be backwards compatible)

Are you having issues with the new compile server? Or are you running your own?

The old servers were flaky with their stability, the new ones should be a lot better

MrWollenburg, 2021-04-23:
> Hey,
>
> With a change in protocol, the old servers from chromeduino 2 are not compatible with version 3 (new v3 servers should be backwards compatible)
>
> Are you having issues with the new compile [...]

Makes sense! I was using old v2 servers. I'll wait for the potential 50% issue to get fixed.  :)  Keep up the great work!

mrfrase3, 2021-04-24:
Thanks! Will do! :+1:

closed 2021-04-24 by mrfrase3