duinoapp/duinoapp-client #25: continued library issues

oldgrizz, 2021-04-25:
cant access libraries or manually add libraries.  example, LiquidCrystal.  I am trying to use chromebooks to program arduino clones outside the arduino ide in an educational setting (jr/high school).  I need to be able to utilize libraries in order to explore other inputs and outputs.  Any help would be great.

mrfrase3, 2021-04-26:
Hey, can you please describe the steps your taking to use the library?

tjbocch813, 2021-05-19:
Okay, so I have run into a couple problems. My biggest problem is getting the program to compile/upload if I am using MULTIPLE libraries.

I have been using the Servo and NewPing libraries and they work fine when I am using just one, but add them both in there and I cannot get anything to compile.

Also make sure that you are only using one library for ALL of the files in your project's folder. Even using different ones in different files have caused problems for me.

This doesn't solve much, but hopefully helps understand the problem.

mrfrase3, 2021-05-19:
That's certainly interesting, I'll have to investigate further. I'll post in here any updates.

stephencunningham4, 2021-06-23:
I can confirm I am also having this exact issue.

ran-to, 2021-07-23:
I can also confirm.

Simplest test to do to replicate this issue:
* add library "Servo", version 1.1.8
* Run the "Check & Compile"
Result: Everything compiles

* add also library "ezTime", or any other library of free choise
* Run the "Check & Compile"
Result: "**0% - Initialising Libraries...**" - and stuck there

It does not matter which order the libraries have been added. the combo of "Servo" library together with others gets stuck.
It is not a problem of number of libraries added either, as adding several other libraries together works fine.

To rule out other issues with libraries, I launched the Arduino Cloud platform instead to test.
There everything compiled with same libraries added.

I also installed my own Duino compiling server with docker on windows machine to test.
Same issue with duino app with that server on localhost compiling.  

Let me know if you would like me to test something more, to figure the bug out on the duino app/server.

Newbie on Arduino stuff and wanted to stay only on my chromebook.
Got really scared getting so much issues early on.
I am not sure how to proceed, but most likely I'll do my coding without this library to the extent I can and then copy code over, add libraries and do final compiling on another cloud platform.

mrfrase3, 2021-07-26:
Interesting, that's a good reproducible flow for the issue, I'll take a look.

mrfrase3, 2021-08-14:
I've fixed this in the latest version, 3.3.1, which is in beta testing at beta.duino.app (make sure you're using the compile-beta compile server)

Let me know if you have any further issues.

closed 2021-12-06 by mrfrase3