duinoapp/duinoapp-client #36: The 'Clear Monitor' button does not always clear the Serial Monitor when using Serial.print method.

jim4101, 2021-06-02:
Conditions:
- Windows 10
- Microsoft Edge and Vivaldi both exhibit anomaly.

Code to reproduce the anomaly:

void setup()
{
  Serial.begin(9600);
  // The Clear Monitor button will erase this line from the Serial Monitor.
  Serial.print("'Clear Monitor' button will clear this line from the Serial Monitor.\n");
}

void setup()
{
  Serial.begin(9600);
  // The Clear Monitor button will not erase this line from the Serial monitor.
  Serial.print("'Clear Monitor' button will not clear this line from the Serial Monitor.\r");
}

mrfrase3, 2021-06-02:
How interesting. I think this is because the clear functionality doesn't clear the current "active" line.

`\r` and `\n` are a carryover from typewriters, `\r` means carriage return (the page moving back to the left) and `\n` means newline (moving the page up so you're on the next line). Normally you always want to do both `\r\n` to the point that most modern programs/systems will accept just `\n`, annoyingly windows is not one of these modern systems and one of the main frustrations that lead developers away from it. Terminals also want both `\r\n` so I have a piece of code that interprets people sending just `\n` and turns it into `\r\n`, but not `\r`, and there are reasons for this...

Technically you could display a number/line of text that overwrites itself using `\r`
```
// assuming your number is never longer than 10 digits
void printNumber(int num) {
  Serial.print("\r          \r");
  Serial.print(num);
}
```
If I auto-corrected a single `\r` into `\r\n`, which would fix the clearing issue, it would also remove the ability to do the above.

hmmm

jim4101, 2021-06-02:
Yep, I'm well aware of the difference in the use of those control characters between Windows and Linux.  Opening a Linux text file in Windows NotePad can be quite ugly.  But what a boring place the world would be without a little diversity to keep things interesting, am I right? :)  I'm even old enough to have used a teletype machine during my amateur radio days, when carriage return and line feed control characters had an actual purpose.

Anyway, rather than scrolling and scrolling and scrolling lots of information on the Serial Monitor I was simply using Serial.print and the C++  \r ANSI escape sequence to place the cursor back to the beginning of the same line and then overwriting the line with new information.  That is when I discovered I could not use the Clear Monitor button to erase the display.  Obviously it is not a show stopping problem, but it was something I found easy to post on GitHub when you get bored with gaming an need a new challenge. :)

I just assumed the Serial Monitor was some type of list or text box control that would have an associated Clear method to erase all the content.  I guess I was thinking that when the Clear Monitor button was selected maybe there was a check to first verify content existed before performing the Clear so maybe a single line of content was not being detected.

Here is the code that caused me to stumble upon this issue...
void reportMeasurement(int measurement, int brightness)
{ 
  Serial.print("\rPhoto Resistor: ");
  Serial.print(measurement);
  Serial.print("     LED Brightness: ");
  Serial.print(brightness);
  Serial.print("        ");
}

jim4101, 2021-06-02:
I just tried the Arduino IDE and discovered the Serial Monitor doesn't interpret the C++ \r ANSI escape code so Serial.print keeps printing to the same line and content is scrolled in the horizontal direction.  The Arduino IDE does recognize the \n escape code.

I'm well aware of the difference in the use of those control characters between Windows and Linux.  Opening a Linux text file in Windows NotePad can be quite disturbing.  But what a boring place the world would be without a little diversity to keep things interesting, am I right? :)

Anyway, since the Arduino IDE ignores the \r escape code I'm not sure this issue is worth your time to investigate since the goal of duino.app seems to be a hosted substitute for the Arduino IDE.  I must admit though, I was disappointed to discover the Arduino IDE Serial Monitor ignored \r since this can be a useful technique to update information on the Serial Monitor without the need to continuously scroll information.  At least I can use the technique with duino.app provided I don't need to clear the Serial Monitor content at some point. :) - Jim

mrfrase3, 2021-06-02:
Thanks Jim

Very interesting. I wonder how the serial monitor handles with both IDEs... I suppose for such an advanced use case it may not be relevant.

I might look into inserting `\r\n` before sending the clear command, that might solve your issue. Either way, the explanation above might be useful for people that stumble upon this issue in the future.

jim4101, 2021-06-03:
From what I have discovered the Arduino IDE (v1.8.13) Serial Monitor recognizes only the \n (newline) and \t (horizontal tab) escape codes.  Duino.app not only supports those to escape codes, but also \b (backspace), \r (carriage return), and \f (formfeed).  Although form feed is a bit odd where the print cursor is moved down one line without changing the position of the print cursor.

I found in the Arduino forum a post where someone was requesting support for the \r escape code.  The user said they were displaying the measurements from a bunch sensors using the Serial Monitor and the scrolling display made it difficult to visually monitor the measurements.  Similar to what I was attempting, the user had a desire to use the \r escape code to overwrite the same line with measurement updates.

In response to the post someone said the Serial Monitor was merely a diagnostic/debug tool and not meant as a terminal application.  It was suggested to use PuTTY instead.  Of course the convenience of using the Serial Monitor, with it being integrated into the IDE, is a huge benefit especially when tweeting and uploading code changes.

Anyway, it does seem odd the Clear Monitor button will not erase the line where the print cursor is positioned.  Moving the cursor to a new line before clearing the serial monitor would appear to be a workaround for the anomaly. - Jim

mrfrase3, 2021-06-03:
Yeh, that sounds like a lazy answer to a legitimate question.

When starting fresh with duino.app, I decided to use [xterm.js](https://xtermjs.org/) which is a very popular library for web-based terminals, so it doesn't surprise me that all that stuff would work out of the box.

+1 point for duino.app I guess

closed 2021-12-06 by mrfrase3