duinoapp/duinoapp-client #38: Serial monitor problem with Arduino board using Silicon Labs CP2102 USB to serial chip

jim4101, 2021-06-06:
I have encounter a problem where the duino.app serial monitor is unable to receive messages after a compile and upload activity.  I have a new Mega board that uses a Silicon Labs CP2102 USB to serial chip and it is this board that exhibits the problem.  My other boards use the ATMEGA 16U2 USB to serial converter.  The new Mega board works okay with the official Arduino IDE.  I decided to put the information in a pdf rather than attempt to document it in a post.  Hopefully this works.

[DuinoBugCP2102.pdf](https://github.com/duinoapp/duinoapp-client/files/6603513/DuinoBugCP2102.pdf)

mrfrase3, 2021-06-11:
hm, this is interesting because I fixed this issue in version 3.2.1, [the fix is in this commit](https://github.com/duinoapp/duinoapp-client/commit/6f754fbb596562546197de66ac1e9058c491f82d)

That fix does wait 5 seconds for the connection to close, I wonder if it's somehow taking long or something :thinking:

jim4101, 2021-06-11:
The problem persists, at least for me, when using a Mega board that has the Silicon Labs CP2102 USB to Serial chip.  The problem does not occur with boards that use the Atmega 16U2 USB to Serial chip.  I haven't done any in-depth analysis to see if I can further isolate the problem, but maybe I should give it a try.  I've been using the workaround of switching the baud rate to another value and switching back, which does resolve the problem until the next code upload.

Just some pondering…

I'm not sure if this could be an issue, but I read in a forum that the Arduino IDE will toggle the serial DTR flow control signal which will hardware reset the Arduino board when the Serial Monitor activated.  I don't know if your code does this somewhere, or if that level of flow control to toggle the DTR signal exists.

Also, unlike the duino.app IDE, the official Arduino IDE will not allow code to upload to the Arduino board while the Serial Monitor window remains open.  I've always had to close the Serial Monitor before uploading code to the board otherwise a com error occurs.

I'm wondering if the Atmega 16U2, being a microcontroller-based USB to Serial converter, performs the DTR reset as a function of the firmware, when the serial port is opened.  Maybe the CP2102, being a hardware solution, acts differently.

It also appears duino.app closes the serial port and will immediately open the port for the Serial Monitor.  I wonder if that process occurs too quickly for the CP2102 to respond, or for the microcontroller reset circuit to respond.  Obviously when I switch baud rates manually that process occurs slowly by comparison.

There is a rather detailed description of the Arduino microcontroller reset process at this [link](https://rheingoldheavy.com/arduino-from-scratch-part-11-atmega328p-dtr-and-reset/).  I probably should read it more thoroughly since it might provide a clue to the cause of the problem.  There is an RC component to the reset circuit so there does seem to be a need to allow a few milliseconds to elapse for the hardware reset to occur.

If you are in the USA I'd be happy to send you one of these CP2102 equipped Mega boards.  I have a feeling I will soon have a surplus of them. :( - Jim

jim4101, 2021-06-12:
Using the Mega board with the CP2102 USB to Serial converter chip, I dusted off my old oscilloscope and took a look at the hardware reset waveforms created by the Arduino IDE and the Duino.app IDE.  I wish I had one of those nice modern digital scopes, but unfortunately mine is an old Tektronics analog model so I had to improvise before I could capture what was happening.

Anyway, what I found is the reset waveforms between the two IDEs were identical so I doubt that could be the problem.  The one difference between the two IDEs is the Arduino IDE creates only one reset pulse at the time of upload.  The Duino.app also creates one pulse at the time of upload, but exactly one second later another reset pulse is created.  This second pulse appears to happen after the upload completes.

Since Duino.app switches the serial port into Serial Monitor mode soon after the upload completes, I wonder if the CP2102 is unable to respond to the change as quickly as needed.  As I mentioned in my original PDF, I do find it odd that the CP2102 is able to send messages from the Serial Monitor, it just ceases to receive messages after receiving just a few characters.

As I also mentioned, the CP2102 does work fine with the official Arduino IDE.  Of course that IDE does not open the Serial Monitor after the upload completes.  The Serial Monitor is manually activated by the user.  I also noticed both the Duino.app and the official Arduino IDE will trigger a reset pulse each time the Serial Monitor baud rate is changed.

I was digging through the Arduino IDE source code, specifically SerialUploader.java and Serial.java.  SerialUploader has a lot going-on in that code, with a noticeable use of the Thread.Sleep method and elapsed time counters, apparently making guesses as to how long to pause execution waiting for external stuff to happen.  While the code does contain some comments, I wish it was more descriptive. - Jim

mrfrase3, 2021-06-12:
Maybe a configuration option to set the timeout could fix it, or just adding an arbitrary wait before reconnecting the device.

Side note: old analog scoped have high enough refresh rates to display images in music, I made a simulation of this effect when you map the left and right audio channels to x and y on the scope. http://dev.3mr.fr/5scope/

jim4101, 2021-06-13:
That scope application is quite interesting and impressive, although, I was surprised the application pushed all the cores of my Intel i7 processor to 50%!  I think you have provided a better CPU stress test than Prime95. :)

In the mid 80s when I purchased my Tektronics analog scope there was an mp3 player option available, but I didn't see a need for it.  After watching your application I now see I missed out on decades of lissajous entertainment. :)

I was surprised to learn you are way down-under and way far-over.  Seeing that the duino.app compile server is hosted in Ohio it was my hunch you were here in the US and maybe even nearby sharing in the same miserable heat and humidity as me.  Now I see you are actually twelve hours ahead of me, living with a rather comfortable climate I might add.

Anyway, I spent a portion of my Saturday researching the Arduino Optiboot bootloader in an attempt to get a clue to the CP2102 serial problem.  I was able to find two decent resources that described the Arduino bootloader process, [here](https://github.com/Optiboot/optiboot/wiki/HowOptibootWorks) and [here](https://arduino.stackexchange.com/questions/15936/what-happens-when-code-is-uploaded-using-the-bootloader/).  Unfortunately I still can't determine what duino.app might be doing that allows it to work okay with my boards using the Atmega16U2 USB chip and not the one with the CP2102 chip.

I believe you have a beta duino.app site so maybe it would be worth a try to add some extra delay after the upload completes, but before opening the serial port for the Serial Monitor.  I assume this change would be easy to add.  Certainly the Ardiuno IDE uses lots of delays to work around external issues.  I just prefer to understand problems such as this a bit more rather than adding code patches and hoping for the best.

I'm left to wonder if opening the serial port for the Serial Monitor, which will trigger an external reset shortly after the code upload completes, if this could be competing with the timeout of the microcontroller watchdog timer set by the bootloader that will also reset the microcontroller.  Obviously both USB chips experience the same condition, but maybe the CP2102 is unable to effectively manage these competing operating state changes.

Probably the best approach for the duino.app Serial Monitor would be to emulate what the Arduino IDE Serial Monitor does.  Open the Serial Monitor serial port only when the Serial Monitor tab is selected, which will reset the microcontroller and restart the sketch.  Then when another tab is selected, say the Program tab, close the Serial Monitor serial port.  I could understand why some people may not prefer this approach, but this does appear to be how the official Arduino IDE manages the serial port for use with the Serial Monitor. - Jim