duinoapp/duinoapp-client #55: 100% Error When Uploading to Device

amknana, 2022-05-15:
Hello,

I am using Arduino Uno and running duino.app on Windows 10. I am trying to code how to sensor temperature and humidity using DTH 11 with Adafruit sensor library.

However when I try uploading the code, it always shows **100% - Error**. Any tips on how to fix this? The code does not show any error.
This is what is shows me: [image]

`#include "DHT.h"
#define DHTPIN 2
#define DHTTYPE DHT11

DHT dht(DHTPIN, DHTTYPE);

void setup() {
  Serial.begin(9600);
  dht.begin(); // initialize the sensor
}

void loop() {
  // wait a few seconds between measurements.
  delay(2000);

  // read humidity
  float humi  = dht.readHumidity();
  // read temperature as Celsius
  float tempC = dht.readTemperature();
  // read temperature as Fahrenheit
  float tempF = dht.readTemperature(true);

  // check if any reads failed
  if (isnan(humi) || isnan(tempC) || isnan(tempF)) {
    Serial.println("Failed to read from DHT sensor!");
  } else {
    Serial.print("Humidity: ");
    Serial.print(humi);
    Serial.print("%");

    Serial.print("  |  "); 

    Serial.print("Temperature: ");
    Serial.print(tempC);
    Serial.print("°C ~ ");
    Serial.print(tempF);
    Serial.println("°F");
  }
}`

Thank you!

mwb9aa, 2022-08-29:
I have the same issue, but with a different code. Still Windows 10, using the new edge browser based on chromium.

amknana, 2022-08-30:
Hi! It's been a long time since I got this problem but I ended up solving
it. Unfortunately as I was just using duino app that one time, I can't
recall how I solved it. :( But we did it using the same device.

If I'm not mistaken, it's related to us clicking the wrong button so it's
just not uploaded right. Maybe the best thing is to really see those who
have done it and see them do it step by step.

mrfrase3, 2022-08-30:
Hey, sorry I didn't get back to you on this one.

If you can take a look at your console logs, it'll tell me more about the issue.

[here's a video on how to do it](https://cdn.discordapp.com/attachments/769521076203683854/812518883725672478/Peek_2021-02-20_10-58.webm)

mwb9aa, 2022-08-30:
Unfortunately, these are student chromebooks, and our district IT has disabled developer tools, including the console log. Some students worked, and some didn't. Class ended before I had a chance to test if the issue could be related to the caching old/bad programming code/different projects, etc.

mwb9aa, 2022-08-31:
I just tried my computer with an Elegoo v. 4.0 robot and software I downloaded from their website. I'm getting a 413 error, which is supposed to be the code is larger than the server can handle.

[http://69.195.111.207/tutorial-download/?t=RobotCarV4.0]

[duino.app-1661973222429.log](https://github.com/duinoapp/duinoapp-client/files/9464355/duino.app-1661973222429.log)

mrfrase3, 2022-09-01:
A simple increase in the request size limit will do the trick.

The default was 100kB, while the `ArduinoJson-v6.11.1.h` file was 176kB alone, it should probably be increased anyway, 100kB is a bit useless...

Give it a go (in like 30 min when it's deployed) and let me know.

closed 2022-09-01 by mrfrase3

mwb9aa, 2022-09-02:
I tried yesterday, and the 413 error is gone. Thanks for your help.