duinoapp/duinoapp-client #61: Does not connect to board

amschmidt2, 2022-10-06:
When trying to connect to the server it does not offer the board to connect to, so the code cannot compile or upload

tongardia, 2022-10-07:
I have the same problem. When i try to connect, i can't select any board at the list, because is empty and there's no possibility to compile and upload the code. I think the problem is related to chrome version.  My chrome version is 106.0.5249.103 (official compilation) (64 bits) on a windows laptop. Yesterday i tried on a linux with a chrome versión 90 and it worked!

I need to use it over chromebook platform, and i can't downgrade  chrome version to a previous version.  

Thanks in advance!

mrfrase3, 2022-10-07:
At the beginning of every month, I do a fresh build of the compile server. The one that just happened on the 1st of Oct, 2022 had a random error so the build was borked. I just ran a new build and it all seems fine now, if you can confirm that would be great.

If people experience this in the future, a short-term fix can be to use the beta server: https://compile-beta.duino.app/
Obviously, the long-term fix is to ping me about it, either here or on discord.