duinoapp/duinoapp-client #67: esp32 no board definitions?

tonygrud, 2023-05-27:
Do the esp32 boards work?  I always get board def errors?
invalid fqbn

mrfrase3, 2023-05-27:
There's an issue with changing away from the default settings in the board selector with ESP's.

Really you should only need to change the clock frequency, which you can do by picking a similar board with the frequency you want.

tonygrud, 2023-05-28:
Hi, thanks for your reply.  I really like your duino app, well done.  I was able to flash my esp32 dev module in the end.  The other issue I found is duino app doesn't detect the webserial disconnect and still thinks it has serial connection and compiles ok but then can't upload.  The disconnect was showing in the browser console though.  So a browser flush after serial disconnect allows it to compile and flash.  I am a school technology teacher and have a small part time coding and robotics business under development.  We have pretty much finished our JS coding editor and the scratch blocks editor that allows students to code both robots over wifi and make scratch games etc.  Now I want to add in arduino coding so excited to find your work because ours is all out of browser so that it can work on ipads/pc or phones etc.  Our site is https://scratchlinkapp.au/ and I'm wondering what your thoughts are about us maybe utilising some of your work?  I would only set it up so that it compiles only to the esp32 dev chip that is in our educational wifi robots rather than a generic thing.  I'm not the programmer but have a partner who does that as I just do all the educational coding stuff which is much easier than the server code.
Cheers
Tony G.