duinoapp/duinoapp-client #68: Mouse issue

Scratch-gu, 2023-08-19:
I am working on an autoclicker and I noiticed there was an error when I tried to upload it. There are some screenshots of the code and error 
[image]
[image]

mrfrase3, 2023-08-19:
Not entirely certain if this is causing the issue, but the Arduino mega cant act as a HID device. You need something with a ATmega32u4 chip, i.e. a pro micro or leonardo.

More annoyingly, the 32u4 chips require the AVR109 uploader that isn't supported on this version of the application.

Scratch-gu, 2023-08-20:
> Not entirely certain if this is causing the issue, but the Arduino mega cant act as a HID device. You need something with a ATmega32u4 chip, i.e. a pro micro or leonardo.
> 
> More annoyingly, the 32u4 chips require the AVR109 uploader that isn't supported on this version of the application.

That might be the issue

closed 2023-08-24 by Scratch-gu