duinoapp/duinoapp-client #69: just get 100%-error

jlrw, 2023-10-19:
I upload and that's what I get. No indication of what the error might be or where it is in the process. It compiles, then I'm told there's an error. Any help appreciated.

nateabele, 2024-09-16:
This either means the board connection was lost, or the board is incompatible. Unplug the device, refresh the page, plug it back in and choose 'Select Device' again. If it still doesn't work, you may have a bad board / third-party Chinese board with incompatible firmware.

mememememememememememememem, 2024-10-25:
> This either means the board connection was lost, or the board is incompatible. Unplug the device, refresh the page, plug it back in and choose 'Select Device' again. If it still doesn't work, you may have a bad board / third-party Chinese board with incompatible firmware.

i will try