duinoapp/duinoapp-client #74: Won't compile (100% - Error!)

onepointfive-REAL, 2025-03-18:
It starts at 25% - Compiling code... then goes straight to 100% - Error!

can anyone help

onepointfive-REAL, 2025-03-18:
hmm looks like library error:
Using board 'uno' from platform in folder: /mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.6
Using core 'arduino' from platform in folder: /mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.6
Detecting libraries used...
/mnt/duino-data/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/mnt/duino-data/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard /tmp/arduino-sketch-FEC26927919244B25FBF722F0C6CB46A/sketch/screen.ino.cpp -o /dev/null
Alternatives for LiquidCrystal_I2C.h: [ ]
ResolveLibrary(LiquidCrystal_I2C.h)
  -> candidates: [ ]
/tmp/duino-1VjilxfXQmidn/screen/screen.ino:1:10: fatal error: LiquidCrystal_I2C.h: No such file or directory
 #include <LiquidCrystal_I2C.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Error during build: exit status 1

mrfrase3, 2025-03-18:
Just sanity checking, you have added this library in the libraries area?