duinoapp/duinoapp-client #76: SSL Certificate Expired?

MrUniat, 2025-05-02:
Good Evening!

I have tried on several laptops and they all say the SSL Certificate has expired for the duino.app site so they cannot access the site. I have managed to connect using an old (technically expired) certificate. Not sure if anyone else is having this issue. The certificate states it expired at the end of March. Could it be renewed if that is the issue?

Thank you!

Dewain7, 2025-05-02:
I, too, can access the site on any device I'd accessed from before, but new devices all get the same SSL error.
Is it possible to alter some info somewhere in new machines to match whatever the old machines have so they can work just the same?

mrfrase3, 2025-05-03:
Thanks for reporting this. There was some kind of new issue that cropped up with the Github Pages config that caused it to stop issuing new certs for the domain.

I've re-connected the domain and it seems to be working now, you may need to refresh for a bit if your browser has cached the old cert.